tenstorrent/tt-umd #769: PCI harvesting not working on P100

broskoTT, 2025-04-17:
As reported by @abhullar-tt :
Im running
./build/test/tt_metal/unit_tests_device --gtest_filter=*DeviceInitializeAndTeardown/0*
 on yyz-lab-125 with branch abhullar/bh-virtual and seeing:
libc++abi: terminating due to uncaught exception of type std::runtime_error: P100 card should always have PCIE core (2, 0) harvested.

This happens on p100. There's probably a bug in logic in https://github.com/tenstorrent/tt-umd/pull/742

nxuTT, 2025-04-21:
Here is a harvesting cheatsheet, it outlines what get harvested for which product: https://tenstorrent.atlassian.net/wiki/spaces/syseng/pages/638615610/Product+Harvesting+Cheatsheet

pjanevskiTT, 2025-04-22:
closed by #772

closed 2025-04-22 by pjanevskiTT