tenstorrent/tt-umd #770: Verify multiprocess support with real processes

broskoTT, 2025-04-17:
Try to spin up a long lived monitor which ocassionally reads something or telemetry, and try spinning up tt_metal tests in parallel, even multiple tests.

Also try to spin up tests which target different chips and see what happens

broskoTT, 2025-04-23:
Today's test:
I ran modified WormholeTelemetry::BasicWormholeTelemetry in such a way to read telemetry in an infinite while loop with sleep(1 sec).
Parallel to it I run the whole suite of ./build/test/tt_metal/unit_tests_dispatch
This was done on a single N150 card.

Waiting on @Joseph to provide additional scenarios to double check if they work.

@adifrancescoTT what is the exact need that you have? Is this task blocking something?

adifrancescoTT, 2025-04-24:
Thanks @broskoTT ! No mostly wanting to get a sense of the status as it relates to Joseph's workstreams, nothing blocked specifically. Thanks again for the prompt update

broskoTT, 2025-05-13:
This was done as an initial test. Some follow up might be available in https://github.com/tenstorrent/tt-umd/issues/848

closed 2025-05-13 by broskoTT