tenstorrent/tt-umd #773: Different types of P100 board use differnet PCI core

pjanevskiTT, 2025-04-22:
As described in #769 , different Blackhole P100 boards are using different PCI cores. Figure out the proper logic to support this, either by introducing both board types or some other additional information.

pjanevskiTT, 2025-04-23:
Just want to point out that broken functionality was covered by #769. This issue is supposed to track scrappy and P100a mapping to certain PCI core, since we only have P100 type as board type currently in UMD. Ideally if #769 fixed metal issues this should become a low priority issue

broskoTT, 2025-07-10:
No action anymore. 
If p100 cards on scrapy boards come up again, we can revisit. But this has been addressed in related PRs

closed 2025-07-10 by broskoTT