tenstorrent/tt-umd #975: Rename Soft Reset variables

nhuang-tt, 2025-06-11:
`tt_metal/third_party/umd/device/api/umd/device/tt_silicon_driver_common.hpp`

Would be nice to rename `TensixSoftResetOptions` or add new values to reflect the reset values for Ethernet cores.

Currently on ethernet, to deassert reset DM1 we need to use
```cpp
TENSIX_DEASSERT_SOFT_RESET ~std::underlying_type<TensixSoftResetOptions>::type(TensixSoftResetOptions::TRISC0));
```

TRISC0 is being used because it's bit 12

broskoTT, 2025-07-21:
@nbuncicTT any opinion on this one?

nbuncicTT, 2025-07-22:
@broskoTT @nhuang-tt Yes, I think that it would be good to add another `enum class` similar `TensixSoftResetOptions`, something like `EthernetSoftResetOptions` and define the bits for the core and use that `enum class` for the assert/deassert of ETH cores.

broskoTT, 2025-07-24:
Feel free to refactor as you see fit

broskoTT, 2025-09-16:
According to https://yyzc-reports.local.tenstorrent.com/bh_a0_reg_map/risc_debug.htm#SOFT_RESET_0.432

I'll be adding enums to the ongoing PR https://github.com/tenstorrent/tt-umd/pull/1241 for ERISC0 and ERISC1 (first one matches BRISC, second matches TRISC0 on tensix)