tenstorrent/tt-umd #1054: Warm reset

broskoTT, 2025-07-07:
Support warm reset through UMD.
By a popular request from metal and exalens

pjanevskiTT, 2025-07-07:
pasting comments from slack

Yes I have a few usecases that all center around the same idea:
I'm a dev, working on kernels, I know I could be hacking something that might hang.  I'd much rather have the system just throw an error + the currently running kernel to yell at me, "you hung the device!!", vs waiting, seeing something's probably hung, shoot yeah it's hung, go to my other terminal window, run tt-smi, wait to make sure it retrained, make a change and repeat.
I'm a cloud user that wants to play with ttnn and I spin up a jupyter notebook and I'm developing a model.  For some reason, my instance is hanging:
Did the server crap out?
Did my wifi stop working?
Oh maybe the cloud system is timed out?
Oh the device is hung hmm
ssh to my cloud instance and run tt-smi reset
It'd be much better if I just got an exception that told me the device hung and then it auto-reset things into a working state.
I'm a developer that wants to deploy a vllm instance using TTNN.  I discover that 1 in 1000 runs hang the device, it'd be much much much better if I could just file a bug and make my script robust and handle a hang exception and retry instead of having to manage multiprocessing on the side.

pjanevskiTT, 2025-09-02:
@broskoTT @nbuncicTT anything else to be done here?