tenstorrent/tt-umd #1071: Enable two flaky RISC deassert tests on Blackhole

pjanevskiTT, 2025-07-09:
Reenable two flaky tests on Blackhole for RISC reset deassert, which are disabled in #1070

broskoTT, 2025-07-21:
I think all of us can't figure out the ordering implications of whole host CPU -> PCI -> TLB -> AXI -> NOC chain.

nbuncicTT, 2025-08-04:
In principle, I have a solution for flaky Blackhole tests:

- Add reading of all locations that were written to

- Add l1 membar between two reads when the counter is running

I discussed this l1 membar approach with a @broskoTT  and in principle it doesn't solve the root cause of the problem, but the test passes likely due to timing.

Problems with this approach:

- It's essentially a solution for the tests, but doesn't solve the fundamental issue of why two consecutive reads get mixed up.

- It's very slow when reading all memory locations that were written to, which would significantly slow down tests (for all RISC core combinations) - I don't know if it even makes sense to implement this change because of a significant performance impact on the CI. 

This branch ([nbuncic/restore_flaky_blackhole_test](https://github.com/tenstorrent/tt-umd/tree/nbuncic/restore_flaky_blackhole_test)) implements the workaround and here are CI results for 5 successful consecutive tests on blackhole (https://github.com/tenstorrent/tt-umd/actions/runs/16591505495).

closed 2026-09-15 by broskoTT