tenstorrent/tt-umd #1092: Collapse PCI device info logs

pjanevskiTT, 2025-07-15:
Collapse logs about PCI device info into few lines

[image]

into something like

```bash
Opened PCI devices [0, 1, ..., 31]
KMD version: 2.0
IOMMU: enabled/disabled
FW version: 18.5.0
```

Relevant slack thread if needed  https://tenstorrent.slack.com/archives/C07L16Z59AP/p1752582528273669

broskoTT, 2025-07-16:
The existing logs should be removed, or changed to log_debug. The log_info showing the same information should be added to Cluster.
Similar pattern can be followed as for the `void Cluster::verify_fw_bundle_version()` in https://github.com/tenstorrent/tt-umd/pull/999/files

pjanevskiTT, 2025-07-25:
closed by #1105

closed 2025-07-25 by pjanevskiTT