tenstorrent/tt-umd #1153: TT_VISIBLE_DEVICES env variable

pjanevskiTT, 2025-08-04:
**Timeline**
No specific timeline

**Is your feature request related to a problem? Please describe.**

https://github.com/tenstorrent/tt-metal/issues/25860
Could help avoid bugs like this in the future

**Proposed solution**

Implement TT_VISIBLE_DEVICES env var. This env var should serve the same purpose as TT_METAL_VISIBLE_DEVICES. It should replace metal env var and should be used both by metal/umd and other runtime components. This is better solution that existing pci_target_devices API because metal has bunch of places where they call enumeration functions from PCIDevice class and somewhere they forget to use TT_METAL_VISIBLE_DEVICES which leads to problems describes in the issue above

pjanevskiTT, 2025-08-11:
@bgoelTT @cfjchu merged the changes. @cfjchu Let me know if any help is needed when tt-metal wants to integrate this

pjanevskiTT, 2025-08-11:
Closed by #1160

closed 2025-08-11 by pjanevskiTT

bgoelTT, 2025-08-11:
Thank you @pjanevskiTT , we don't have an issue in tt-metal for integrating this feature right? If not, I will create one

cfjchu, 2025-08-12:
Yes, we need your help to integrate the changes through metal - thanks!

bgoelTT, 2025-08-12:
I created an issue here https://github.com/tenstorrent/tt-metal/issues/26658, please feel free to add any more detail you feel necessary in the comments