tenstorrent/tt-umd #1237: Fix simulator and host connection PR

pjanevskiTT, 2025-08-29:
https://github.com/tenstorrent/tt-umd/pull/1236#pullrequestreview-3169124504 reverted a commit that changed communication between host and simulator because tt-sim was failing

Check in this commit again with proper fix for both VCS and tt-sim

broskoTT, 2025-08-30:
This change breaks our tt-umd-simulators repo.

@tt-vjovanovic can you help @mcraigheadTT implement the new protocol in tt-sim?

mcraigheadTT, 2025-08-30:
I'm confused, why doesn't the pipe work?  I would prefer to not switch to socket, as this is extra work, could cause various potential issues (e.g. may regress performance), would require significant testing, and doesn't strike me as the right long term direction.  I believe the right long term direction is to get rid of the IPC.

mcraigheadTT, 2025-08-30:
I haven't gotten around to it yet, but my plan this upcoming week was to remove the IPC altogether for ttsim and have everything run in 1 process.

mcraigheadTT, 2025-08-30:
I suppose we could still support an IPC mode in ttsim after that, though I'm not convinced that it is worth the trouble to continue supporting it.  I do know that the current IPC is very, very slow

broskoTT, 2025-09-03:
The fix is in UMD, so both codepaths work. 
Further development on this topic will be tracked by https://github.com/tenstorrent/tt-umd/issues/1254